jonas-schievink/rubble #13: Panic after running for 1-5 seconds

chocol4te, 2019-03-08:
```
Panicked at 'called `Result::unwrap()` on an `Err` value: Eof', src/libcore/result.rs:997:5

Program received signal SIGTRAP, Trace/breakpoint trap.
0x00002e54 in __bkpt ()
    at /Users/ferdiamckeogh/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.7/src/lib.rs:454
454	    }
```

I've narrowed it down to the `.unwrap()` on line 161 of `src/ble/bytes.rs`. `T::from_bytes(b)` is returning an end of buffer error.

jonas-schievink, 2019-03-08:
Current theory is that the radio is modifying the buffer behind our back, even though it should be idle. This is clearly undefined behaviour as well, so very important to fix.

jonas-schievink, 2019-03-17:
Should now be fixed on master!

closed 2019-03-17 by jonas-schievink