jonas-schievink/rubble #25: Support thumbv6 target (again)

jonas-schievink, 2019-03-28:
Thumbv6 support was (accidentally) dropped when pulling in bbqueue (also see https://github.com/jamesmunns/bbqueue/issues/25). We need to fix that issue or stop using bbqueue, and then add a thumbv6 build job to CI.

This is needed for nRF51 support, since those all have a Cortex-M0.

chocol4te, 2019-03-28:
I went the trouble of bumping the log PR, all for nothing 😢

jonas-schievink, 2019-03-28:
Yeah, sorry for not noticing this. That log PR was still needed though, we just need to do a bit *more* work now. This isn't very critical to do since we're on nRF52s anyway, but the easiest fix for now is probably to use the stuff in [`heapless::spsc`](https://docs.rs/heapless/0.4.2/heapless/spsc/index.html) instead of `bbqueue`.

jonas-schievink, 2019-11-29:
This was done in https://github.com/jonas-schievink/rubble/pull/97, but I believe hasn't been fully tested yet. I'll still close this, any problems can be filed as follow-up issues.

Please help test this if you have the time <3

closed 2019-11-29 by jonas-schievink