jonas-schievink/rubble #34: Design a crypto interface

jonas-schievink, 2019-04-02:
For encrypted connections, the platform will provide a hardware-implementation of the needed crypto algorithms (mainly, AES-CCM). We have to incorporate that into our hardware interface. This will probably involve changing `Transmitter` or creating a subtrait of it to be implemented by radios with encryption support.

Things to consider:

* Nordic devices can do (and recommend doing) encryption and transmission in a single pass. Can we incorporate this or should we just not support it?
* Not all platforms might offer crypto hardware, so this must stay optional (hence the subtrait idea).
* How does the crypto hardware look across different manufacturers?

ryankurte, 2019-06-14:
hi hi, y'all might be interested in checking out https://github.com/rust-embedded/embedded-hal/issues/142, definitely wouldn't want to be a blocker but if it's possible to design something consistent that'd be super neat for stack portability ^_^

jonas-schievink, 2019-06-14:
@ryankurte Thanks, I've left a comment there. Looks like there wasn't a lot of progress since I last investigated, unfortunately, but we can always define BLE-specific traits until the general ones mature.