jonas-schievink/rubble #38: Use company identifiers in LL_VERSION_IND

chocol4te, 2019-04-03:
Currently a `Hex<u16>` is used to represent `CompId`, this should be an enum of company identifiers which implements `.as_u16()`.

The full list can be found here:
https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers

jonas-schievink, 2019-04-03:
Alternatively it could also be a `struct CompanyId(u16);` with associated constants for the companies. I'm not sure which is better yet (the enum would need an additional `Unknown` variant carrying the raw `u16`, and also a `#[doc(hidden)]` variant for extensibility since `#[non_exhaustive]` is not yet stable).

In any case it should also get a `.name()` function that returns the exact company identifier from the bluetooth.com list above.

A `fmt::Display` implementation would also be neat.

chocol4te, 2019-04-07:
Started work on #44

jonas-schievink, 2019-04-07:
You can put "Closes #38" in your PR description and it'll close this issue automatically when merged, no need to do it before the fix is actually merged.

chocol4te, 2019-04-07:
Ah sorry, TIL

PR #44 (2019-04-08): feat: company id working
Closes #38
closed 2019-04-08 by jonas-schievink