jonas-schievink/rubble #48: Design a better hardware/radio interface

jonas-schievink, 2019-04-21:
The current interface relies on undocumented assumptions and isn't very clean. It should be fixed before we start supporting more MCUs to avoid introducing too much technical debt.

(the timer interface is basically fine at this point, this issue is about the radio interface)

For example, the `transmit_advertising` and `transmit_data` methods on the `Transmitter` trait behave *very* differently since `transmit_data` also ensures `T_IFS` is respected. This isn't really wrong, but it should *also* be done by `transmit_advertising`, *except* when we're just a beacon. The solution for this, I think, is to split the `Transmitter` trait into "beacon-only" and "connection" functionality:

* Beacon-only use requires being able to do single-shot transmission and reception on the advertising channels (eg. no TX->RX sequence that respects `T_IFS`). It's very simple to implement.
* Connections require *only* TX->RX sequences that respect `T_IFS`, on both advertising and data channels. It is more difficult to implement.

Another problem @chocol4te has faced is that the current `BleRadio` implementation for the nRF52810 *only* works with a proper `LinkLayer`, and doesn't allow plugging in a `BeaconScanner`. This is ultimately an interface decision made by the device crate, but maybe we can find a solution in Rubble itself (perhaps via some sort of `PacketSink` trait?) that makes this easier.

ryankurte, 2019-06-14:
ooh, interesting problem ^_^

i don't have enough of a grasp of BLE to make suggestions atm, but i do have another compatible [radio-sx128x](https://github.com/ryankurte/rust-radio-sx128x) i'd be interested in implementing against and some experiments with  [radio](https://github.com/ryankurte/rust-radio) interfaces if those are useful references.

jonas-schievink, 2019-06-14:
@ryankurte That's a very interesting chip! I was looking for something like it. Can you recommend a board that uses it (preferably one that doesn't cost [450€](https://www.mouser.de/ProductDetail/Semtech/SX1280DVK1ZHP?qs=5aG0NVq1C4zPNIv0R5By4w%3D%3D))?

I'll also keep an eye on your radio crate. It does seem to be lower-level than what we're looking for here, but could still be helpful nonetheless.

ryankurte, 2019-06-14:
> That's a very interesting chip! I was looking for something like it. Can you recommend a board that uses it (preferably one that doesn't cost 450€)?

i've been using the [DLP-RFS1280](https://www.digikey.co.nz/product-detail/en/dlp-design-inc/DLP-RFS1280/813-1052-ND/8038152) modules, which are still not cheap but workable for prototyping.

> It does seem to be lower-level than what we're looking for here, but could still be helpful nonetheless.

yeah it might well be too low level, i'm planning to push that up a bit with [nb](https://japaric.github.io/nb/nb/) implementations next i have the time, but, definitely interesting to see what works for BLE. when y'all have a `Receiver` trait too i'll have a go at implementing it on these ^_^

ryankurte, 2019-06-14:
the radio hal is also v much an experiment atm, because i've suffered through too many frustratingly implemented radio drivers in the past, so i'm totally up for modifying it to suit if it's reasonable to do so ^_^

eldruin, 2019-06-15:
You can also get one of [these modules](https://www.aliexpress.com/wholesale?SearchText=sx1280) for 7-10 USD.