jonas-schievink/rubble #51: Make rubble-nrf52810 support all nRF52 chips

jonas-schievink, 2019-04-25:
The nRF52 chips are very similar (except the nRF52840, which has additional features for ZigBee and Thread - it should still work similarly though), so we should try to make the existing `rubble-nrf52810` crate work with all of them.

The best way to do this is probably by renaming `rubble-nrf52810` to `rubble-nrf52` and then using Cargo features and `#[cfg]` to select a chip, similar to what the `nrf52-hal-common` crate does.

chocol4te, 2019-04-26:
Would now also be the time to make a `rubble-nrf51` crate?

jonas-schievink, 2019-04-26:
That requires https://github.com/jonas-schievink/rubble/issues/25, so the crate wouldn't compile right now

chocol4te, 2019-04-26:
🤦‍♂️ I forgot about that :/

jonas-schievink, 2019-04-28:
Fixed by https://github.com/jonas-schievink/rubble/pull/54

closed 2019-04-28 by jonas-schievink