jonas-schievink/rubble #56: iOS not seeing ~50% of nRF52810 modules

chocol4te, 2019-04-28:
On 2 out of 4 tested chips, when running the example code, the device is not visible to iOS (tested on iPhone 7 and XS). All chips work with Android devices. Currently believed to be due to inter-frame spacing issues during scan events.

jonas-schievink, 2020-10-25:
Note that this issue only affects *scan events*, which are used when a device is sending *connectable* (or scannable) advertising PDUs. It does not affect beacons, which are *non*-connectable advertising PDUs.

dthoma, 2020-10-30:
I've got the same issue on Android (Pixel 3a).
If I change the code to send a non-connectable PDU, Android does find the device, but it fails to for a connectable PDU. I can see the scan requests in the trace. From my laptop the device works just fine.

unrelentingtech, 2021-07-23:
Trying the characteristic demo on an nRF51822 (obviously with slight modifications to run on 51), Android does not discover it at all while my desktop (FreeBSD with Intel Bluetooth) does.

nebelgrau77, 2021-09-27:
BLE n00b here. The demo works fine for me, but it's only visible to my kubuntu machine with an EDUP brand Bluetooth/WiFi dongle. Neither Rasperry Pi 4, nor two Android phones with nRF Connect can see it. I tried it with two different boards, Arduino Nano BLE Sense (u-blox chip) and Adafruit ItsyBitsy (Raytec chip). 
So... how do I change it to send a non-connectable PDU, then? 

BTW no luck with the beacon demo, no device can see it.

unrelentingtech, 2021-09-27:
> how do I change it to send a non-connectable PDU, then?

Doing what beacon does.

> no luck with the beacon demo, no device can see it

Very weird, that's with the interfaces that show everything (like nRF Connect), not just various OS built-in device scanners that only show connectable devices?

nebelgrau77, 2021-09-28:
> Doing what beacon does.
> 
which would be what exactly? broadcast instead of advertise? 

> Very weird, that's with the interfaces that show everything (like nRF Connect), not just various OS built-in device scanners that only show connectable devices?

yes, nRF Connect on two different Samsung Axx phones, as well as bluetoothctl on the Linux machine, the one that can see the other demo.