jonas-schievink/rubble #63: Disconnect handling

chocol4te, 2019-06-19:
Rubble should gracefully disconnect and return to regular broadcasting when a client sends a disconnect request.

jonas-schievink, 2019-06-19:
We already handle this and return back to the "idle" Link-Layer state (which is demanded by the spec). We should have some sort of interface for notifying a user that this has happened though.

chocol4te, 2019-06-19:
What causes the `missed conn event`?

jonas-schievink, 2019-06-19:
That gets printed when we're waiting for a connection event but don't receive the data packet of the master in time. Usually we can recover from that by hopping and waiting for the next one. Are you seeing an endless stream of that? Then the connection has been lost, which shouldn't normally happen (unless you've forcefully disconnected / turned off the other device, or the connection is *actually* lost due to a weak signal or something).

One thing that *is* still missing is the connection supervision timer, which would terminate *lost* connections in the same way as voluntary disconnects after a configured timeout.

chocol4te, 2019-06-19:
Ahh okay, I think I might be having another iOS-specific issue. I get the stream even when disconnecting properly.

jonas-schievink, 2019-06-19:
It isn't strictly required to send the "disconnect pls" request, so iOS might just drop the connection instead of properly closing it. What do you see before the `missed conn event` errors start?

chocol4te, 2019-06-19:
To me it looks pretty instantaneous after pressing "Disconnect" in the nRF Connect app :/

jonas-schievink, 2019-06-19:
(I mean, beside the missing supervision timer, of course)

chocol4te, 2019-06-19:
:)

closed 2019-06-19 by chocol4te