jonas-schievink/rubble #71: bbqueue is used incorrectly

jonas-schievink, 2019-08-09:
As discovered in https://github.com/jonas-schievink/rubble/pull/64, bbqueue's `grant_max` doesn't work as documented (https://github.com/jamesmunns/bbqueue/issues/29), so our usage of it is wrong and can lead to packet processing stalling.

This either needs a documentation fix and additional methods in bbqueue, or we should switch to a different queue. Ideally, that queue would also support thumbv6 while still having efficient batch operations.

jonas-schievink, 2019-08-30:
8b62569 uses a fork that includes https://github.com/jamesmunns/bbqueue/pull/33. We should move back to the upstream repo or a crates.io release as soon as we can.

jamesmunns, 2019-11-29:
@jonas-schievink is this something that would be fixed by https://github.com/jamesmunns/bbqueue/pull/37?

Or do you need one of the proposed grants at https://github.com/jamesmunns/bbqueue/issues/38, maybe `grant_largest` or `grant_largest_max`?

jonas-schievink, 2019-11-29:
@jamesmunns Does [`grant_max_remaining`](https://docs.rs/bbqueue/0.4.0-alpha5/bbqueue/atomic/struct.Producer.html#method.grant_max_remaining) now work as documented? If so then that's all I need.

jamesmunns, 2019-11-29:
Ah no, good catch! It works as documented at the module level (see https://docs.rs/bbqueue/0.4.0-alpha5/bbqueue/atomic/struct.Producer.html), but it is a little vague at the function level, which I will be fixing. For reference:

* `grant_max_remaining(N)`:
  * User will receive a grant `0 < sz <= N` (or receive an error)
  * This will only cause a wrap to the beginning of the ring if exactly

I *think* you actually want `grant_largest_max()`, which will cause a wraparound if the requested space is NOT available at the end, but IS available at the beginning.
      zero bytes are available at the end of the ring.
  * Maximum possible waste due to skipping: 0 bytes

jamesmunns, 2019-11-29:
I need to come up with a consistent naming scheme, but in general:

* `remaining` methods do NOT cause premature wraps
* `largest` methods MAY cause premature wraps
* `max` methods limit the upper bound of grantable size

jonas-schievink, 2019-11-29:
> I _think_ you actually want `grant_largest_max()`, which will cause a wraparound if the requested space is NOT available at the end, but IS available at the beginning.

The only remaining use of bbqueue is in the logging utilities, which *should* be fine with getting a small buffer first, and then wrap around and get another buffer. It will only "fail" (it will consider data to be lost) if the function returns `Err`.

jamesmunns, 2019-11-29:
I also plan to offer split grants in the future, which will give you the buf at the start and end, so you can decide at once if there is enough space, but I haven't put any detailed design or dev into that, planning on doing that post 0.4.0.

jonas-schievink, 2019-12-08:
Fixed by #100

closed 2019-12-08 by jonas-schievink