jonas-schievink/rubble #74: LLCP updates are not applied when the event is missed

jonas-schievink, 2019-08-30:
When we've stored an LLCP update, we need to apply it to the connection state after the correct connection event number (or "instant"). Currently, this is done here:

https://github.com/jonas-schievink/rubble/blob/71be7a1bfa596ffa0024be64257bb4165c12036c/rubble/src/link/connection.rs#L276-L281

However, this code is inside the RX handler, so if we happen to miss the event we're supposed to apply the update, we never do so, resulting in the connection being lost completely. This can be seen in the following log:

```
2.888649s - TRACE - #18 DATA(15->22)<- Header { LLID: DataCont, NESN: 1, SN: 1, MD: false, Length: 0 }, []
2.889144s - INFO - LLCP<- ConnectionUpdateReq(ConnectionUpdateData { win_size: 2, win_offset: 17, interval: 39, latency: 0, timeout: 500, instant: 29 })
...
2.963649s - TRACE - #28 DATA(11->18)<- Header { LLID: DataCont, NESN: 1, SN: 1, MD: false, Length: 0 }, []
2.964147s - TRACE - DATA(18->25): missed conn event #29
2.972340s - TRACE - DATA(25->32): missed conn event #30
2.980533s - TRACE - DATA(32->2): missed conn event #31
2.988725s - TRACE - DATA(2->16): missed conn event #32
2.996940s - TRACE - DATA(16->16): missed conn event #33
...
```

Event 29 is missed presumably due to interference, we never apply the update, all future events are missed because the connection state is now desynced.

jonas-schievink, 2019-08-31:
A good way to fix this would probably be to move the code that hops channels and applies the LLCP update into its own function (`close_connection_event`?) and just call it from the 2 places.

lulf, 2020-12-04:
I'm seeing this on my nRF51 as well. I'm trying to apply the change you suggested @jonas-schievink, but I'm not sure if I'm doing it in the correct place. I basically took line 271-283, removed the 'queued_work' stuff, and patched it in timer_update().