jonas-schievink/rubble #96: master fails to compile with const_or issue

JJJollyjim, 2019-11-27:
I'm banging my head against a wall trying to figure out what I'm doing wrong here:

compiling the latest master, on an up to date stable or nightly compiler, I get:

```rust
~/P/r/rubble-nrf52 ❯❯❯ cargo +nightly check --features="52810" --target "thumbv7em-none-eabi"                                  ✘ 101 
    Checking rubble v0.0.3 (/home/jamie/Projects/rubble/rubble)
error[E0015]: calls in constants are limited to constant functions, tuple structs and tuple variants
  --> rubble/src/gatt/characteristic.rs:24:54
   |
24 |         <const_or!(@[$($t)+])>::from_bits_truncate($(($t :: $bit).bits())|+)
   |                                                      ^^^^^^^^^^^^^^^^^^^
...
56 |     const PROPS: Properties = const_or!(Properties::READ | Properties::WRITE);
   |                               ----------------------------------------------- in this macro invocation

error[E0015]: calls in constants are limited to constant functions, tuple structs and tuple variants
  --> rubble/src/gatt/characteristic.rs:24:9
   |
24 |         <const_or!(@[$($t)+])>::from_bits_truncate($(($t :: $bit).bits())|+)
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
...
56 |     const PROPS: Properties = const_or!(Properties::READ | Properties::WRITE);
   |                               ----------------------------------------------- in this macro invocation

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0015`.
error: could not compile `rubble`.

To learn more, run the command again with --verbose.
```

This seems like it would have failed CI if something was wrong, so I'm certain I'm doing something questionable, but I can't figure out what.

jonas-schievink, 2019-11-27:
Oh, that's odd. Can you post the output of `rustc +nightly -Vv`? I'm on `rustc 1.41.0-nightly (797fd9262 2019-11-26)` and it's building fine.

JJJollyjim, 2019-11-27:
Exact same:

```
rustc 1.41.0-nightly (797fd9262 2019-11-26)
binary: rustc
commit-hash: 797fd92628842c1f5face9fb93b0fe4f1f9d297f
commit-date: 2019-11-26
host: x86_64-unknown-linux-gnu
release: 1.41.0-nightly
LLVM version: 9.0
```

jonas-schievink, 2019-11-27:
Try deleting `Cargo.lock`, maybe you have an outdated `bitflags` crate

JJJollyjim, 2019-11-27:
Oh of course, thanks so much :)

commit 2615aa8e (2019-11-27): Require newer version of bitflags

1.0.x doesn't provide `const fn`s, which we need.

Fixes #96
closed 2019-11-27 by JJJollyjim