jonas-schievink/rubble #103: Encrypted connection unsupported?

loccs-faker, 2020-03-16:
Normally, after connection established, a session key will be generated to encrypt(AES-128) follow communication. 

It seems that the encrypted connection is not supported by rubble now.(related LL control PDUs, like LL_ENC_REQ and LL_ENC_RSP, are not implemented. ) 

Is encrypted connection in the plan of this project?

jonas-schievink, 2020-03-16:
Yeah, right now Rubble doesn't support encryption. The plan is to eventually support it, but that will still take a lot of work that's being tracked in https://github.com/jonas-schievink/rubble/issues/34. https://github.com/jonas-schievink/rubble/issues/48 is also relevant since encryption is most often provided by the radio hardware itself. And of course we need support for pairing as well, which depends on https://github.com/jonas-schievink/rubble/pull/102 getting merged so that we can do public key cryptography on P-256.

jonas-schievink, 2020-03-16:
If you'd like to help out with this, I think the most useful thing to do would be to sketch out a design for the radio interface that properly supports radios that do support encryption, and radios that don't, ideally in a type-safe manner.

The first step would probably be to write down operations that a radio with encryption support would have to support (Is it just "enable encryption with this AES key" and "disable encryption" or does implementing the spec require more? How should a failing MIC be indicated to the stack?).

jonas-schievink, 2020-07-19:
Closing in favor of #34 and https://github.com/jonas-schievink/rubble/issues/36

closed 2020-07-19 by jonas-schievink