jonas-schievink/rubble #105: Any recommended hardware for getting involved with helping out?

twitchyliquid64, 2020-04-25:
Hi!

First off, awesome project!!! :grin:  I want to get ble / wireless-DFU working on the PineTime, so I'm keen to help out with rubble however I can.

Would you have any recommendations for what hardware to get? Preferably something with USB ([maybe this](https://www.adafruit.com/product/4481)?) so I dont have to solder some dodgy SWD wiredoodles?

And for debugging, are yous using a BLE sniffer or anything fancy like that?

jonas-schievink, 2020-04-25:
Hi!

Ideally you'll want to get a board with a built-in programmer (since the nRF52840 has a USB peripheral, just looking for a USB connector might get you the wrong thing).

The official Nordic dev boards have an on-board J-Link, so they should work. I'm currently using the [nRF52840-MDK](https://github.com/makerdiary/nrf52840-mdk) by Makerdiary, which also does the job and has an on-board DAPLink programmer (though I did have to update the programmer firmware first).

If you're fine with messing around with bootloaders, then a board without a programmer might also work for you, but I don't really recommend that since it's usually a pain to get working.

Another option might be something like the [Particle Xenon](https://www.adafruit.com/product/3999) or [other boards](https://www.adafruit.com/product/4062) with a debugging connector. Then you can use a normal cable to connect a debug probe.

Something else to look out for is the serial connection. Rubble relies a lot on logging via the UART, so if you can get a decent connection to your computer (at at least 115 kbaud, though 1 Mbaud is better) then you'll have an easier time.

I don't currently use a BLE sniffer, but I think Nordic has a sniffer firmware for the nRF51, so that might be useful if you have some boards on hand. I usually rely on the logging output and Wireshark for more high-level protocols.

dbrgn, 2020-06-05:
I have the [Ebyte E73-TBB](http://www.ebyte.com/en/product-view-news.aspx?id=889) (evalboard for the [E73-2G4M04S1A](http://www.ebyte.com/en/product-view-news.aspx?id=587)) and I quite like it so far! You can get it on Aliexpress.

kuon, 2020-09-10:
I have a laird dev kit and it has everything on board:

https://www.mouser.ch/new/laird-connectivity/laird-bl654-kits/