jonas-schievink/rubble #134: Constructing Attribute requires costructing private HexSlice type

daboross, 2020-07-23:
Hi!

It looks like `rubble::att::Attribute`, https://jonas-schievink.github.io/rubble/rubble/att/struct.Attribute.html, has the value data wrapped in a `rubble::utils::HexSlice`. But as the docs indicate, `HexSlice` isn't public - `utils` isn't public, and `HexSlice` isn't re-exported. Thus, as far as I can tell, it isn't possible to construct an `Attribute` from outside the crate?

If this is a problem, do you it'd be reasonable to just re-export `HexSlice` from the crate root, or would it be better to add a constructor and accessors to `Attribute` which allow interacting with value as an `&[u8]`?

jonas-schievink, 2020-07-23:
I'd rather not expose `HexSlice`, so adding a constructor sounds good. Ideally the user wouldn't even have to deal with raw `Attribute`s (instead only with Characteristics and Services), but we aren't there yet.

daboross, 2020-08-05:
Sounds good to me! Submitted #139. And again, my apologies for communicating in short bursts with periods of radio silence.

PR #139 (2020-08-10): Create constructor and value-getter for Attribute::value
Fixes #134! This adds an `Attribute::new` function taking `&[u8]` and `Attribute::value` function returning the value as `&[u8]`.

I've made this as a minimal change - I think it could be reasonable to add getters for the other fields for consistency, and/or make the fields private, but those aren't strictly necessary to fix this issue.

Let me know if there's anything else I can do with this!
closed 2020-08-10 by jonas-schievink