jonas-schievink/rubble #135: Why BleRadio requires static lifetimes for its buffers?

dzervas, 2020-08-01:
https://github.com/jonas-schievink/rubble/blob/17bb8bd292bfe644f85aa185358ff9107df9fd2d/rubble-nrf5x/src/radio.rs#L74-L80

I'm not any kind of a guru in Rust and I'm trying to get rubble to play without `rtic` (btw: it works! :tada: ). The only hacky thing I had to do was this:

```rust
	static mut BLE_TX_BUF: PacketBuffer = [0; MIN_PDU_BUF];
	static mut BLE_RX_BUF: PacketBuffer = [0; MIN_PDU_BUF];
	let mut radio = unsafe { BleRadio::new(
		board.RADIO,
		&board.FICR,
		&mut BLE_TX_BUF,
		&mut BLE_RX_BUF,
	) };
```

I don't get why BleRadio would require `'static` - you need the references to be alive as long BleRadio instance is alive - no?

If there's no particular reason, do you want me to open an MR?

BTW: GREAT, FUCKING GREAT work, thank you for your project

jonas-schievink, 2020-08-01:
The buffers need to have `'static` lifetime because the radio uses DMA to access them. This is somewhat tricky to handle correctly, since the `BleRadio` can be dropped or leaked while the radio still has access to the buffer, so it might write to deallocated memory if we don't require a `'static` lifetime there.

dzervas, 2020-08-02:
hmm, ok, that covers my question. I guess it's a difficult problem to solve - if I do happen to solve it, I'll make an MR.

closed 2020-08-02 by dzervas