jonas-schievink/rubble #145: External code can't access Consume fields, and thus can't implement Consumer

daboross, 2020-08-13:
I ran into this when copying over the implementation from #137 into a crate outside of `rubble` - it looks like `Consume`, the type used for conveying results to `Consumer`, has private fields only accessible to rubble. Unless I'm missing a way to get into `Consume::consume` and `Consume::result`, this prevents outside users from implementing queues built to their needs.

I think this could be fixed pretty easily by either making fields public (maybe with an adding private `()` field for forward compatibility), or by adding accessors. If it's alright, I'll submit a PR - I mildly favor accessors, but since rubble's still pre-`1.0`, and `Consume` is fairly feature-complete for what it is, it probably doesn't matter a ton?

jonas-schievink, 2020-08-13:
Oof, yeah, feel free to add accessors.

daboross, 2020-08-13:
Will do then. Filed #148!

PR #148 (2020-08-14): Add accessors to Consume
Fixes #145

I haven't thought a ton about the naming of these methods, but they should work alright.
closed 2020-08-14 by jonas-schievink