jonas-schievink/rubble #150: Allow to mutate readable attributes

daschl, 2020-10-08:
Hi,

I've been implementing my own `AttributeProvider` and I am running to an (what I think is) a limitation. But maybe there is already way to deal with this?

Right now the `BleChannelMap` takes ownership of my `AttributeProvider` but I need to get to it again through a mutable reference so I can update attributes as my program goes on. For example in RTIC I update a resource (i.e. a sensor temp) and then I need to propagate this into the `AttributeProvider` so that it is reflected when the property is fetched next time.

I wonder if somewhere on the `Responder` and deeper down into the `BleChannelMap` it should be possible to get a mutable reference on the `AttributeProvider` so that I can use setters to update my attributes?

If this is right now not possible I'm happy to contribute it, whatever way you guys think is the most efficient one.

Thanks!

jonas-schievink, 2020-10-08:
Yup, this isn't supported at the moment. In general we need to come up with a better GATT interface (like #72). I'm not yet sure what the best way to do that is.

daschl, 2020-10-08:
For now would it be okay to add `&mut` accessor so I can access it this way?

daschl, 2020-10-09:
Since my pr is merger I‘m going to close this ticket. We are going to revisit it with the rework I assume

closed 2020-10-09 by daschl