jonas-schievink/rubble #174: Getting started

0b-code, 2021-01-19:
Apologies for the noob questions I am fairly new to embedded stuff and having a gdb question. Trying to run some of the demos here, after copying to .gdbinit, get to gdb prompt, but unsure how to proceed. Looking at docs says should do something like target  remote :3333, but target not found. Try using serial target, but get timeouts. Any help appreciated. Thanks!

jonas-schievink, 2021-01-19:
Hi! I highly recommend working through the [Discovery book](https://rust-embedded.github.io/discovery/) if you're new to embedded. It goes through the process of connecting GDB to your target device via OpenOCD.

Bluetooth LE is fairly complex, and rubble isn't very user-friendly yet, so starting with the discovery book should be much more enjoyable.

0b-code, 2021-01-20:
Hey thanks for responding so quickly! Have gone through a few projects that are a bit more sophisticated than your link and OpenOCD/GDB usually work out of the box (if they are used at all).  BLE may be complex, but this repo is fairly easy to follow, which is awesome. The only issue is that the demo does not work, which means debugging is more difficult.

0b-code, 2021-01-20:
Is it possible target not supported? Working from nrf52833, which does have a feature flag, but is not listed as supported in the readme

jonas-schievink, 2021-01-20:
It should work in theory, though I've never tried it. I would expect this to work like any other nRF project for your hardware.

0b-code, 2021-01-20:
So to be clear, intended behavior is that beacon should not launch immediately, but should go through GDB? Usually workflow is `openocd -f openocd.cfg` to `(gdb) target remote :3333` or `(gdb) /some/serial_target` and `load` `continue` etc. From your latest PR seems like you might be willing to recommend a rust-native workflow, which would be awesome too

0b-code, 2021-01-20:
To clarify edit, `probe-run --chip nRF52833_xxAA` appears to work with beacon, but not visible.

0b-code, 2021-01-21:
Seems like this may have to do with Micro:bit BLE profile, which is a bit frustrating since this could be solid recommended hardware board. Not closing yet though since still should be able to work

0b-code, 2021-01-25:
Closing and confirming this works on Micro:bit and nrf52833!

closed 2021-01-25 by 0b-code