jonas-schievink/rubble #187: the trait `NrfTimerExt` is not implemented for `nrf52840_hal::nrf52840_pac::TIMER0` error

alsoft-alex, 2021-11-06:
Hello,

When compiling my application, I get the following error:

 _let ble_timer = BleTimer::init(device_peripherals.TIMER0);
   |                                        ^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `NrfTimerExt` is not implemented for `nrf52840_hal::nrf52840_pac::TIMER0`
   |
note: required by `BleTimer::<T>::init`
  --> /Users/xxxx/.cargo/registry/src/github.com-1ecc6299db9ec823/rubble-nrf5x-0.0.4/src/timer.rs:38:5
   |
38 |     pub fn init(mut peripheral: T) -> Self {
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0277]: the trait bound `nrf52840_hal::nrf52840_pac::TIMER0: NrfTimerExt` is not satisfied
  --> src/main.rs:52:25
   |
52 |         let ble_timer = BleTimer::init(device_peripherals.TIMER0);
   |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `NrfTimerExt` is not implemented for `nrf52840_hal::nrf52840_pac::TIMER0`
   |
  ::: /Users/xxxx/.cargo/registry/src/github.com-1ecc6299db9ec823/rubble-nrf5x-0.0.4/src/timer.rs:30:24
   |
30 | pub struct BleTimer<T: NrfTimerExt> {
   |                        ----------- required by this bound in `BleTimer`_

Here are my dependencies:
_[dependencies]
rubble = "0.0.4"
rubble-nrf5x = "0.0.4"
cortex-m = "0.7.3"
cortex-m-rt = "0.7.0"
cortex-m-rtic = "0.5.9"
cortex-m-semihosting = "0.3.7"
panic-halt = "0.2.0"
rtt-target = { version = "0.3.1", features = ["cortex-m"] }
nrf52840-hal = { version = "0.14", features = ["rt"], optional = true }

[features]
52840 = ["rubble-nrf5x/52840", "nrf52840-hal"]_

I compile the app by calling cargo build --features 52840.

Does anybody have any idea what to do?

danielzfranklin, 2022-01-15:
(not the maintainer) I suspect your problem is that `NrfTimerExt` is implemented for a different version of `nrf52840_pac::TIMER0` than your application code is using. I solved this by using the unpublished version of rubble, you could also figure out what version of `nrf52840-hal` rubble uses and downgrade to that version in your app.

Here's how to use the latest unpublished version

```toml
rubble = { git = "https://github.com/jonas-schievink/rubble", features = [
    "log"
] }
rubble-nrf5x = { git = "https://github.com/jonas-schievink/rubble", features = [
    "52840"
] }
```

alsoft-alex, 2022-01-25:
Thanks @danielzfranklin !